SpinalHDL/SpinalDoc-RTD #3: License change

wifasoi, 2018-12-21:
Hi, sorry if i mention you in this issue, but we need you to approve the license change to from MIT to CC0 (for this fork).
If you're ok with this please reply, otherwise we can discuss this here.
@lambdaShade 
@pmudry 
@Snoopy87 
@Dolu1990 (pro-forma)

Thank you for your time.
Luca Cristaldi(wifasoi)

Dolu1990, 2018-12-21:
So, the idea about making it CC0 is to make it as easy as possible to reuse and share.

numero-744, 2022-08-24:
Has the license been changed?

RTD indicates "© Copyright 2022, SpinalHDL" and there is no LICENCE file in the root of the repository

Dolu1990, 2022-08-25:
i think the license didn't changed, but we can go CC0

wifasoi, 2022-08-25:
Unfortunally no one responded. I can add the CCO licence now.. and it will be truw for all new contribution

Dolu1990, 2022-08-25:
>  I can add the CCO licence now.. and it will be true for all new contribution.

Ok to me :)

numero-744, 2022-10-02:
Can be closed as done in 6e4d40382409764f10966849410fe1b996a5dfbf

closed 2022-10-02 by wifasoi