SpinalHDL/SpinalDoc-RTD #5: Have a TODO list

wifasoi, 2019-01-02:
I think have a todo list of what is missing in the documentation is handy.

philipaconrad, 2020-08-27:
Here's the TODO list I've built up while working on the docs over the last week or two. I had a pile of sticky notes on my desk, and figured it'd be wise to get into digital form before I lose anything. :sweat_smile: 

### Additions to existing pages

 - [ ] Add `Vec[Reg[T]]` example to the [Vec][rtd-vec] page. This would illustrate that registered types can be stuffed into `Vec`s just as easily as their combinational variants. (This was non-obvious to me when I first started using Spinal, as there were no examples in the docs of it.)
 - [ ] Add `Vec[Type]` function parameter examples on the [Function][rtd-function] page. This could make it easier for Scala newbies to get the type signatures correct on medium-complexity functions.
 - [ ] Add a `Component` renaming example to the [Component and hierarchy][rtd-component-and-hierarchy] page. This is a useful feature especially when working in legacy RTL projects, where fixed naming schemes may be expected. Currently, this functionality is only documented on the [Blackboxing (Instantiate VHDL and Verilog IP)][rtd-blackboxing] page.
- [ ] Updates to the [Users][rtd-users] page. The "repos" section especially could use some love, and if we could get links to the websites of companies using Spinal, that'd be awesome too.

### Global changes

 - [ ] More consistent code-formatting in the API sections across the data-types pages. Currently, some pages have better code formatting around their API docs than others. Having these be more consistent would be nice. :smile: 

### Proposed new pages

 - [ ] `MaskedLiteral` / `MaskedBoolean` docs. These are mentioned in a footnote on the [Data types][rtd-datatypes] page, but are not comprehensively covered anywhere else in the docs.
 - [ ] An "Internals" or "Built-ins" page documenting `spinal.core` built-ins, like `exp` / `bits` / `bits` / `pos` / `slice`, etc. This might live under the [Developers Area][rtd-developers-area] section.

### Proposed new code examples

 - [ ] Examples of `reduce` / `reduceBalancedTree`. These are *very* useful (espec. for things like adder trees), and don't seem to be comprehensively shown off in examples, or documented thoroughly anywhere. (This could be worth doing in a "Functional Programming Idioms" section or something. :man_shrugging: )
 - [ ] Pipelining examples. These examples would demonstrate how to implement pipelining for a simple ALU or other component, and might cover idiomatic usage for `Reg` / `RegInit` / `RegNext` when used in chains of registers.
 - [ ] Data structure examples. These examples would demonstrate how to use Scala collections to make generating certain types of hardware much easier. (Example: specifying a [sorting network][w-sorting-network] as a list of swap indices.)
 - More `Memory` -related examples! `Memory` types, especially RAMs, seem to have sparse documentation overall. Example ideas might include:
    - Loading a RAM's initial contents from a hex file, altering some values, and reading the updated values.
    - Putting a RAM in as a data source for an ALU or something.
    - Using a RAM/ROM inside an existing component. (IIRC, [SpinalCrypto][gh-spinalcrypto] may have some examples of this with lookup tables, [S-Boxes][w-s-box], etc.) 

   [rtd-vec]: https://spinalhdl.github.io/SpinalDoc-RTD/SpinalHDL/Data%20types/Vec.html
   [rtd-function]: https://spinalhdl.github.io/SpinalDoc-RTD/SpinalHDL/Structuring/function.html
   [rtd-datatypes]: https://spinalhdl.github.io/SpinalDoc-RTD/SpinalHDL/Data%20types/index.html
   [rtd-component-and-hierarchy]: https://spinalhdl.github.io/SpinalDoc-RTD/SpinalHDL/Structuring/components_hierarchy.html
   [rtd-blackboxing]: https://spinalhdl.github.io/SpinalDoc-RTD/SpinalHDL/Structuring/blackbox.html
   [rtd-users]: https://spinalhdl.github.io/SpinalDoc-RTD/SpinalHDL/About%20SpinalHDL/users.html
   [rtd-developers-area]: https://spinalhdl.github.io/SpinalDoc-RTD/SpinalHDL/Developers%20area/index.html
   [w-sorting-network]: https://en.wikipedia.org/wiki/Sorting_network
   [w-s-box]: https://en.wikipedia.org/wiki/S-box
   [gh-spinalcrypto]: https://github.com/SpinalHDL/SpinalCrypto

Dolu1990, 2020-08-27:
Nice :D
Also, the simulation API documentation is far from complet. Quit a few things were added since the last doc update ^^

philipaconrad, 2020-09-01:
The simulation API is something I found pretty intimidating to work with. It's definitely on my list of sections/pages to look into. :sweat_smile:

Dolu1990, 2020-09-01:
@philipaconrad don't hesitate to ask question about it, i'm often on the gitter channel ^^
So the simulation API is made of very simple primitives which compose well together, just a mater to get used to :)

numero-744, 2022-08-24:
I think the TODO list should be the issues list: seems way more easy to deal with it than with a single issue with all the stuff as checkboxes, and having to add items to it + with a really long discussion mixing all topics.

Btw these TODO lists seem outdated.

Dolu1990, 2022-08-25:
Yes i think that's right, openning issues for individual items is more tractable.

numero-744, 2022-10-02:
Can we close this issue?

Dolu1990, 2022-10-03:
Yes as it is not realy keept up to date, let's try to switch to individual issues.

closed 2022-10-03 by Dolu1990

numero-744, 2022-10-09:
(Maybe it can be unpinned too ^^')