SpinalHDL/SpinalDoc-RTD #6: ScalaDoc hosting

wifasoi, 2019-01-02:
I think having a ScalaDoc page hosted (maibe on SpinalHDL repo) can be handy. Should we add it to the CI/CD flow?

Dolu1990, 2019-01-02:
So instead of hosting on https://spinalhdl.github.io/SpinalDoc-RTD/, hosting directly on https://spinalhdl.github.io/SpinalHDL/ ?

wifasoi, 2019-01-02:
I don't mean this documentation. I mean to have something similar to [this](https://www.scala-lang.org/api/current/scala/collection/parallel/immutable/index.html)

and it easier to have only site per repo (I can use [sbt-site](https://www.scala-sbt.org/sbt-site/index.html), to maintain two site in one gh-pages.. but i prefer to keep it simple and have one gh-paage per repo)

So (to be clear), my idea is to:
* Host/generate the scaladoc documentation on https://spinalhdl.github.io/SpinalHDL/ (SpinalHDL repo)
* Keep the high-level documentation here https://spinalhdl.github.io/SpinalDoc-RTD/ (SpinalDoc-RTD repo)

Dolu1990, 2019-01-02:
Ahhh right. i got it now, Yes i agree, and it was already requested by other peoples :)

Dolu1990, 2019-01-02:
I will try to do that.

wifasoi, 2019-01-16:
SpinalHDL scaladoc is hosted here: http://spinalhdl.github.io/SpinalHDL/
so, I'll close this issue.

closed 2019-01-16 by wifasoi

piegamesde, 2019-12-02:
Please link it somewhere on <spinalhdl.github.io>. I had to find this issue to get a link to the ScalaDoc, which is not ideal.