SpinalHDL/SpinalDoc-RTD #12: Investigate image conversion error

wifasoi, 2019-01-17:
A commit introduce a error
``` text
WARNING: convert exited with error:
[stderr]
Unit… specificata non valida. (italian for Specified unit not valid)
```
this is between 2ba70f7 (bad, acutal dev HEAD) and d161f14 (good, master HEAD 1.3.0 tag)
Tote to self: run git bisect and find the commit responsible for this

numero-744, 2022-08-24:
Any updates about this issue?

If the error still exists, please open an issue in https://github.com/SpinalHDL/SpinalHDL/issues

Else this issue can be closed.

Dolu1990, 2022-08-25:
> If the error still exists, please open an issue in https://github.com/SpinalHDL/SpinalHDL/issues

But it is a SpinalDoc-RTD issue isn't it ?

numero-744, 2022-08-25:
Yes sorry, my bad.

I did not understand the error message was from a docs generator and not SpinalHDL or Scala :disappointed:

Dolu1990, 2022-08-25:
No worries ^^

closed 2022-08-25 by Dolu1990