SpinalHDL/SpinalDoc-RTD #35: Where to submit PRs while reading

jonnew, 2020-05-05:
Great job with this project. 

As I read the docs, I'm encountering a lot of grammatical errors. My question is: is it worth it for me to correct these errors as I read and issue PRs? If so, on which branch? I'm noticing that there are many features in the code that are not present in the docs and I don't want to put the effort in if the Docs are going stale, so to speak. Unfortunately, I don't have the required expertise to bring the docs up to date with the code, yet, but figured English proof-reading might be useful.

Dolu1990, 2020-05-05:
Hi :)

So, basicaly, all the doc about things which are already released in a SpinalHDL version can be directly added in the master branch, while if that's something which isn't yet released, it can go to the dev branch.

This doc need more love ^^, i agree it is a bit stall, but additions are more than welcome.

So go ahead as much as you would like / enjoy :)

philipaconrad, 2020-08-24:
@jonnew I saw some small grammatical errors that were bothering me, and just started making copyediting PRs for them. The maintainers are usually pretty quick to review and act on PRs, which is refreshing after some projects I've worked with.

The [rst](https://www.sphinx-doc.org/en/master/usage/restructuredtext/basics.html) format is pretty straightforward to edit, so if there's a page that's really bothering you, it could take about 5-10 minutes to make a PR for, if you've already got a fork to work with.

My workflow has been:
 - Make edits
 - `git commit`
 - `git push origin master` to push the changes to my fork
 - File a PR
 - (After a PR is merged) `git fetch upstream` -> `git rebase upstream/master` -> `git push origin master` to update my fork to track the main docs repo.

andreasWallner, 2021-02-22:
@philipaconrad For small edits like grammar typos I often use the online-edit features on github, makes the process much quicker - and the risk for breaking the docs are most of the time minimal...

numero-744, 2022-08-24:
This question has answers, I think the issue can be closed now.

closed 2022-08-25 by Dolu1990