SpinalHDL/SpinalDoc-RTD #52: Spelling mistakes

liuwei9, 2021-01-28:
In the (https://github.com/SpinalHDL/SpinalDoc-RTD/blob/master/source/SpinalHDL/Structuring/function.rst')
You need to use the case class to define the MyBus class, and you should use def when you override the AsMaster method.

For example
~~~
case class MyBus(payloadWidth: Int) extends Bundle with IMasterSlave {
    val valid   = Bool
    val ready   = Bool
    val payload = Bits(payloadWidth bits)

    // Define the direction of the data in a master mode
    override def asMaster(): Unit = {
        out(valid, payload)
        in(ready)
    }
 // Connect that to this
    def <<(that: MyBus): Unit = {
        this.valid   := that.valid
        that.ready   := this.ready
        this.payload := that.payload
    }

    // Connect this to the FIFO input, return the fifo output
    def queue(size: Int): MyBus = {
        val fifo = new MyBusFifo(payloadWidth, size)
        fifo.io.push << this
        return fifo.io.pop
    }
}
~~~

wifasoi, 2021-01-28:
The case class is not an error (per se), but in the following lines it the class is initialized like one. So your remarks are correct
(and using case class is better anyway) so I'll put in the pile of "will fix later when I have a sliver of free time".
(If you want have a jab at it I can instruct how to contribute to the documentation :D)

Anyway, thanks for spotting it.

wifasoi, 2021-01-28:
haha, too slow *with finger guns* :P

commit 7f16d28b (2021-01-28): Fix example mistakes

fix #52
closed 2021-01-28 by wifasoi