SpinalHDL/SpinalDoc-RTD #87: update the bus_slave_factory

hanm2019, 2022-03-31:
It seems that the [bus salve factory doc](https://github.com/SpinalHDL/SpinalDoc-RTD/blob/master/source/SpinalHDL/Libraries/bus_slave_factory.rst) has no update 3 years ago, but the `spinal.lib.bus` had implemented lots of new functions,  such as:`setOnSet`,`readAndSetOnSet`, ect. 

I have made a list about that:
```
createReadAndClearOnSet
readAndClearOnSet
clearOnSet
createReadAndSetOnSet
readAndSetOnSet
setOnSet
createReadMultiWord
createWriteMultiWord
createWriteAndReadMultiWord
multiCycleRead
readSyncMemWordAligned
readSyncMemMultiWord
writeMemMultiWord
writeMemWordAligned
```

It would be better to update the doc to make others use those functions clearly.  :>

Dolu1990, 2022-04-04:
Hi,

Yes, it missed quite a few updates.

Do you feel up for a PR ?

hanm2019, 2022-04-04:
I can try to update some doc, but for some functions , I am not clear :>

Dolu1990, 2022-04-05:
Ahhh, which ones ?