SpinalHDL/SpinalDoc-RTD #96: Need info on TraversableOncePimped

ronan-lashermes, 2022-05-04:
I am preparing pushing a few doc updates, and I would like to add stuff on Vec helper functions accessible through the TraversableOncePimped class. I have a simple question: why are some function names prefixed by "s" and some aren't ?
E.g. sExist, sCount vs shuffle, read.

I am wondering if I am missing something before writing some doc on that.

Dolu1990, 2022-05-04:
There is a few case where scala get confused, seems especialy with landa function parameter if the collection already had a similarly named function with landa returning another type.

One instance would be : 
```scala
    val z = Vec(U(0),U(0)).count(e => e != null) // OK
    val z = Vec(U(0),U(0)).count(e => (e === U(0) )) //KO
    val z = Vec(U(0),U(0)).sCount(e => (e === U(0) )) /OK  
```

So sCount was used to avoid this issue.

Dolu1990, 2022-05-04:
Let's me know if you have any question :)

ronan-lashermes, 2022-05-04:
Thanks, that's the info I was looking for !