SpinalHDL/SpinalDoc-RTD #116: make html failed

hanm2019, 2022-09-12:
python : 3.7.13 (anaconda)

install the package by `pip3 install -r requirements.txt`

output:
```
Running Sphinx v1.8.6

Extension error:
Could not import extension sphinxcontrib.wavedrom (exception: No module named cairosvg)
Makefile:19: recipe for target 'html' failed
make: *** [html] Error 2
```

andreasWallner, 2022-09-13:
That looks like CairoSVG was not correctly installed (as a dependency for spinxcontrib.wavedrom).
Of the top of my head:
- did you encounter any errors on the `pip install`?
- are you sure that the python that is used in the Makefile is the one that you are `pip3` is installing for? (had that one before, and I'm asking because AFAIK pip is always called just `pip` in a venv) I'd check that by first checking the output of `which python3` (that should be in the virtualenv), `which sphinx-build` should point to the same virtualenv. Then running `python3 -m pip list` lists all installed packages, which should include CairoSVG.

andreasWallner, 2022-09-29:
I just realized from the output that your environment tried running Sphinx v1.8.6 - in the requirements file we are stating 4.1.2, so something must have gone wrong during the install of that...

hanm2019, 2022-10-11:
It works! I found the path of `python3` and `sphinx-build` are not the same. I, therefore, removed the `sphinx-build` and reinstalled it. The version of `sphinx-build` is 4.1.2 now.

closed 2022-10-11 by hanm2019