SpinalHDL/SpinalDoc-RTD #131: Add section for idioms

numero-744, 2022-10-24:
Spinal enables representing the code in a way which is not possible in RTL.

This section would show idioms which are useful in Spinal. For each idiom, it would explain it, how to use it, when and why it is good to use it, and maybe how it works.

Example of idiom:

```scala
val doSomethingFlag = False
def doSomething(): Unit = doSomethingFlag := True
```

I find it useful in `fsm.State.whenActive` to express the output logic.

```scala
  val fsm = new StateMachine {
    val IDLE, Rnd0, RndI, RndN, DONE = new State
    setEntry(IDLE)

    // Transition logic
    IDLE.whenIsActive { when(io.start) { goto(Rnd0) } }
    Rnd0.whenIsActive { goto(RndI) }
    RndI.whenIsActive { when(round.isLast) { goto(RndN) } }
    RndN.whenIsActive { goto(DONE) }
    DONE.whenIsActive { when(!io.start) { goto(IDLE) } }

    // Output logic, using only this idiom
    IDLE.whenIsActive { round.resetCount }
    Rnd0.whenIsActive { run; useInputs }
    RndI.whenIsActive { run }
    RndN.whenIsActive { run; withoutMixColumns }
    DONE.whenIsActive { enableOutput }
  }
```

Dolu1990, 2022-10-24:
Yes, such section would be usefull, showing all kind of funny / creative thing

Readon, 2022-10-24:
Nice Work!
This will be a shining pool of sparks for SpinalHDL.

Dolu1990, 2022-10-25:
Maybe we can do a call to discuss about patterns ?

numero-744, 2022-10-25:
I was thinking that each pattern would be introduced by a PR into the which we could discuss but a call might be good too.

Dolu1990, 2022-10-26:
sure.

numero-744, 2022-11-15:
Note for when I will have created this section:

Function-like components: a companion object with an `apply`.

```scala
  object MyComponent {
    def apply(i: SpinalEnumCraft[MyEnum.type]): Bits = {
      val myComponent = new MyComponent
      myComponent.io.i := i
      myComponent.io.o
    }
  }
```

numero-744, 2022-11-17:
Once idioms are defined we can provide snippets (for instance providing them in the template) for IDEs. For instance:

```json
	"Doing flag": {
		"scope": "scala",
		"prefix": "doing",
		"body": ["val ${2:${1}ing} = False", "def $1(): Unit = $2 := True", ""]
	}
```

We can also provide such snippets for `Component` etc. Providing this kind of ready-to-use tools can improve the Scala UX XD

With this example snippet in VSCodium with 8 keystrokes:

```
do<Tab>run<Tab><Tab>
```

I get:

```scala
val runing = False
def run(): Unit = runing := True
```