SpinalHDL/SpinalDoc-RTD #132: Error trying to generate pdf

numero-744, 2022-10-24:
PDF generation is done in two phases: rst->tex & tex->pdf. I get an error in the first phase.

Expected: create latex files

Actual:

```zsh
% make latex
Extension error:
convert exited with error:
[stderr]
b"convert-im6.q16: delegate failed `'rsvg-convert' -o '%o' '%i'' @ error/delegate.c/InvokeDelegate/1966.\nconvert-im6.q16: unable to open file `/tmp/magick-js2pMmJNdpHh0gLO_sYh6hvy9N2dsNJn': No such file or directory @ error/constitute.c/ReadImage/614.\nconvert-im6.q16: no images defined `/docs/docs/doctrees/images/process_rtl.png' @ error/convert.c/ConvertImageCommand/3229.\n"
[stdout]
b''
make: *** [Makefile:19: latex] Error 2
```

Readon, 2022-10-24:
According to the message you give, it seems that some files and directory is missing, which is not met from my environment.

numero-744, 2022-10-24:
> How about add the compiling PDF task to CI, so that it can be easily released without worry about the compilation environment.

This was my next plan :wink: I am first trying to get it locally in a docker image to fix this kind of issues. (Using Dockerfile in `master` branch)

> which is not met from my environment

Good thing! Now I have to understand why it fails…

numero-744, 2022-10-24:
v5.2.0

numero-744, 2022-10-24:
With 4.1.2 I get the Python 3.10 errors

Readon, 2022-10-24:
Have you tried to compile the docs directly by
``` shell
sphinx-build -M latexpdf "source" "docs"
```

numero-744, 2022-10-24:
Same result

Readon, 2022-10-24:
I think the log would show you some clue.

wifasoi, 2022-10-25:
Works on my end.. with sphinx 5.2 and this is the pip freeze. let m
```
alabaster==0.7.12
Babel==2.10.3
cairocffi==1.4.0
CairoSVG==2.5.2
certifi==2022.9.24
cffi==1.15.1
charset-normalizer==2.1.1
cssselect2==0.7.0
defusedxml==0.7.1
docutils==0.16
idna==3.4
imagesize==1.4.1
Jinja2==3.1.2
MarkupSafe==2.1.1
packaging==21.3
Pillow==9.2.0
pycparser==2.21
Pygments==2.13.0
pyparsing==3.0.9
pytz==2022.4
PyYAML==6.0
requests==2.28.1
six==1.16.0
snowballstemmer==2.2.0
Sphinx==5.2.0
sphinx-multiversion==0.2.4
sphinx-rtd-theme==0.5.2
sphinxcontrib-applehelp==1.0.2
sphinxcontrib-devhelp==1.0.2
sphinxcontrib-htmlhelp==2.0.0
sphinxcontrib-jsmath==1.0.1
sphinxcontrib-qthelp==1.0.3
sphinxcontrib-serializinghtml==1.1.5
sphinxcontrib-wavedrom==3.0.2
svgwrite==1.4.3
tinycss2==1.1.1
urllib3==1.26.12
wavedrom==2.0.3.post3
webencodings==0.5.1
xcffib==0.12.1
```

If you have a problem to convert an svg image.. maibe cairoSVG is not installed, or out of date?

wifasoi, 2022-10-25:
Ok i can reporduce it with:
```
FROM sphinxdoc/sphinx

WORKDIR /docs
ADD requirements.txt /docs
RUN export DEBIAN_FRONTEND=noninteractive && apt update && apt install -y latexmk texlive-latex-recommended texlive-fonts-recommended tex-gyre texlive-latex-extra && pip3 install -r requirements.txt
```
and execute the container with:
`docker run --rm -v $(pwd):/docs <imagehash> make latexpdf`

numero-744, 2022-10-25:
Notice that there is also:

```dockerfile
FROM sphinxdoc/sphinx-latexpdf
```

See https://hub.docker.com/r/sphinxdoc/sphinx

numero-744, 2022-10-25:
I tried with the `sphinx-latexpdf` but I got the same error message

wifasoi, 2022-10-25:
Found atleast one error: you need to install `librsvg2-bin`

wifasoi, 2022-10-25:
The in the conf.py you can add `render_using_wavedrompy = True`, or install wavedrom.. I'm lazy and tried only with former, but it's better if we use wavedrom directly.

numero-744, 2022-10-25:
Thanks, I'm adding npm etc to the [Dockerfile](https://github.com/SpinalHDL/SpinalDoc-RTD/blob/master/Dockerfile). Then I'll have to add steps into actions to generate it, and I'm wondering: where do we want the pdf to be so that it can be downloaded?

wifasoi, 2022-10-25:
as reference: https://github.com/Holzhaus/sphinx-multiversion/issues/70
Maybe we can put in the static folder, and customize the version picker with link to the files, or use the PR linked in the discussion above (need to check it first)

numero-744, 2022-10-25:
@wifasoi I blame you: https://github.com/SpinalHDL/SpinalDoc-RTD/blame/883f8ad4e6c35d5c75e089f12e5261fcada35978/source/conf.py#L138

"I'm a big company. Is this project serious?" *generates pdf doc* "Err…*

wifasoi, 2022-10-25:
LOL, guilty as charged :P
I should use the corporate appropriate placeholder: "Miao" :D

numero-744, 2022-10-26:
So I pushed a branch with Docker configuration using npm version of wavedrom, and with a new Dockerfile which makes it possible to build pdf.

https://github.com/Holzhaus/sphinx-multiversion/pull/74 brings the feature we want but it is still open (last activity 1y ago).

A solution can be to use `smv_prebuild` mechanism to provide the feature we want. Will still have to write an `rst` file pointing to the static PDF file, but the link would be broken if built without `sphinx-multiversion` (eg: local build), actually I don't think it is really an issue.

numero-744, 2022-10-26:
I have pushed to the PR, it works for current branch:

[image]

(PDF opens PDF file, HTML is a broken link for now because I have not enabled it, it should not be displayed actually, I'm gonna report it to the author of PR I used)

But for "all tags" it fails because it fails for one of the tags (so it just stops and do not even try subsequent tags):

```
LaTeX Warning: File `{assert/imag/fixpoint/highBitOperation}.png' not found on
input line 4866.
```

numero-744, 2022-11-02:
@wifasoi any idea about last message?

Readon, 2022-11-02:
> I have pushed to the PR, it works for current branch:
> 
> [image]
> 
> (PDF opens PDF file, HTML is a broken link for now because I have not enabled it, it should not be displayed actually, I have reported it to the author of PR)
> 
> But for "all tags" it fails because it fails for one of the tags (so it just stops and do not even try subsequent tags):
> 
> ```
> LaTeX Warning: File `{assert/imag/fixpoint/highBitOperation}.png' not found on
> input line 4866.
> ```
> 
> Note: things to add to the workflow, see https://github.com/sphinx-doc/docker/blob/master/latexpdf/Dockerfile + `pdf.Dockerfile`

good work.

about the warning, if other images work then there should be an "e" in the path, the correct one is "asset/imag**e**/fixpoint/highBitOperation.png".

numero-744, 2022-11-14:
@Dolu1990

Dolu1990, 2022-11-15:
> So am I allowed to change the commit to the which the tags point to, to fix these links?

Ahh sure sure ^^

Dolu1990, 2022-11-15:
> Okay I fixed tag v1.3.8, now multiversion works!

Nice ^^

> By the way I notice we have branches named vX.X.X which are not tags, is it something we want?

i think we can get ride of them.

numero-744, 2022-11-15:
Ok I'll remove these branches later™

numero-744, 2022-11-15:
There is also branch `1.7.3`, do I remove it? @Dolu1990

numero-744, 2022-11-15:
For `dev` and `old_dev` I am wondering, should `dev` be used instead of `master`, like for SpinalHDL repository?

PR #133 (2022-11-17): ci: pdf generation with docker
Fix #132 

I still have to add it to GitHub actions, but there is still discussion needed in #132 about this.
closed 2022-11-17 by Readon