SpinalHDL/SpinalDoc-RTD #137: Document setWhen and clearWhen

numero-744, 2022-11-03:
Should be in: https://spinalhdl.github.io/SpinalDoc-RTD/master/SpinalHDL/Sequential%20logic/registers.html
(source: https://github.com/SpinalHDL/SpinalDoc-RTD/blob/master/source/SpinalHDL/Sequential%20logic/registers.rst)

Simplified example:

```scala
val inFrameLate = Reg(Bool) setWhen(!sclRead) clearWhen(!inFrame)
```

Dolu1990, 2022-11-03:
hmm setWhen isn't only for register. It can also be used for signals

```scala
val myOutput = out(False)
myOutput setWhen(xxxx)
myOutput clearWhen(yyy)
```

numero-744, 2022-11-04:
Hmm, if I understand correctly, in your example for a simple wire, the `clearWhen` is not needed?

Readon, 2022-11-05:
For the priority, as it is coded
``` scala
def setWhen(cond: Bool)(implicit loc: Location): Bool   = { when(cond){ this := True }; this }
def clearWhen(cond: Bool)(implicit loc: Location): Bool = { when(cond){ this := False }; this }
```
Then according to the last win policy, for "setWhen(cond1) clearWhen(cond2)", clearWhen have high priority, as I understand.

Dolu1990, 2022-11-06:
> Or it is to counter the setWhen?

Right

> Is it equivalent to the line below?

Yes

> To me it is a different use for a register and a wire. So in Scaladoc it would be the same documentation, but in RTD (which is more tutorial-like) it would be good to explain how it is useful in different contexts.

Sure, no problem with it ^^

> For a wire it just builds a comb equation?

Not realy equiation, but more as Readon said, a priority based logic.

numero-744, 2022-11-07:
Great, yakafokon (fr, phonetic expression « (il n')y a qu'à…, (il) faut qu'on… » (en: "one just has to…, one has to…") to tell that we defined what to do but there is nobody yet to do it).

Contributions appreciated :pray: :heart: