SpinalHDL/SpinalDoc-RTD #144: Document CombInit

numero-744, 2022-11-16:


Readon, 2022-12-18:
The main issue would be in which case user should use CombInit, in which it is not necessary.

numero-744, 2022-12-18:
I think I address it in the branch I'm working on, that I'll publish as a PR in the next weeks (refactor step 2). I'll let you tell me then :wink:

Readon, 2022-12-18:
> I think I address it in the branch I'm working on, that I'll publish as a PR in the next weeks (refactor step 2). I'll let you tell me then 😉

Good work.

Dolu1990, 2022-12-19:
@Readon To me there is a few main usages : 
- Creating a combinatorial value copy of a signal, that you can then override without impacting the source value
- Creating a combinatorial value copy of something that you can then name as you whish for (debug purpose / visibility)

Readon, 2022-12-19:
> @Readon To me there is a few main usages :
> 
> * Creating a combinatorial value copy of a signal, that you can then override without impacting the source value
> * Creating a combinatorial value copy of something that you can then name as you whish for (debug purpose / visibility)

Like to generate a new wire through existing wire/reg in netlist?

numero-744, 2022-12-19:
> @Readon To me there is a few main usages :
> 
>     * Creating a combinatorial value copy of a signal, that you can then override without impacting the source value
> 
>     * Creating a combinatorial value copy of something that you can then name as you whish for (debug purpose / visibility)

Thanks I'll add that

Dolu1990, 2022-12-20:
> that you can then override without impacting the source value

I would say  : 
that you can then conditionaly override without impacting the source value

> Like to generate a new wire through existing wire/reg in netlist?

Yes

ronan-lashermes, 2023-11-10:
I missed this issue, but there is a short documentation of CombInit in #228.

andreasWallner, 2023-11-10:
Fixed by @ronan-lashermes with #228

closed 2023-11-10 by andreasWallner