SpinalHDL/SpinalDoc-RTD #145: equivalent constructs at elaboration time/hardware/sim

numero-744, 2022-11-16:


numero-744, 2022-11-16:
@Readon

What do you think of this?

https://sublime-and-sphinx-guide.readthedocs.io/en/latest/code_blocks.html#code-examples-in-multiple-languages

Maybe do the same thing for VHDL vs SystemVerilog vs SpinalHDL?

Readon, 2022-11-16:
that would be great if there are SpinalHDL/Verilog/SystemVerilog/VHDL，user can easily compare and study by that.

However, it's necessary to make that happen.

 

 
 
https://sublime-and-sphinx-guide.readthedocs.io/en/latest/code_blocks.html#code-examples-in-multiple-languages

numero-744, 2022-11-16:
> However, it's necessary to make that happen.

@Readon I'm focusing on RTD for the next 2 weeks, we'll see what it gives ^^

I wanted a big rework of RTD, I have started this morning. I'm waiting for other PRs to be merged (automated code checks and pdf generation in GH workflow) before I publish a PR.

Dolu1990, 2022-11-16:
Can be fancy :D

numero-744, 2022-11-17:
Ooops https://github.com/svx/sphinxcontrib-osexample/issues/8