SpinalHDL/SpinalDoc-RTD #146: Copy button for code

numero-744, 2022-11-16:
What do you think about it?

https://sphinx-copybutton.readthedocs.io/en/latest/

Dolu1990, 2022-11-16:
Sure, why not

numero-744, 2022-11-16:
[image]

Hmm does not render as expected XD

numero-744, 2023-05-31:
Interesting answer: https://github.com/executablebooks/sphinx-copybutton/issues/101#issuecomment-1569264918

And accurate:

https://github.com/SpinalHDL/SpinalDoc-RTD/blob/9d000b2bdf58af01b4b9dd4151574f300b35905e/source/conf.py#L198-L202

Suggestions:

- change the way we override the default CSS, like [`executablebooks/sphinx-copybutton/sphinx_copybutton/__init__.py`](
https://github.com/executablebooks/sphinx-copybutton/blob/4a8050dea36536b005d3e757b1ac09886174f634/sphinx_copybutton/__init__.py#L92)

```python
    app.add_css_file("theme_overrides.css")
```

- merge [`copybutton.css`](https://github.com/executablebooks/sphinx-copybutton/blob/master/sphinx_copybutton/_static/copybutton.css) into our custom CSS?

dlmiles, 2023-05-31:
Merge PR #178 (I have already applied the fix there because I also found issues with other things not working as I expected, so all the work I had planned could not be completed, like updating to sphinx-rtd-theme=~latest, but I have updated that now as well in #178)

Then apply the obvious installation of this plugin.

```
diff --git a/requirements.txt b/requirements.txt
index e200aa6c8..6e281fda0 100644
--- a/requirements.txt
+++ b/requirements.txt
@@ -2,3 +2,4 @@ sphinx==5.3.0
 sphinx-rtd-theme==1.2.1
 sphinxcontrib-wavedrom==3.0.4
 sphinx-multiversion @ git+https://github.com/samuel-emrys/sphinx-multiversion.git@cd723351f687c98d32834226ae7b3ec9e63bcba5
+sphinx-copybutton==0.5.2
diff --git a/source/conf.py b/source/conf.py
index 844749d22..3ae0d4c36 100644
--- a/source/conf.py
+++ b/source/conf.py
@@ -45,7 +45,8 @@ extensions = [
     'sphinx.ext.githubpages',
     'sphinxcontrib.wavedrom',
     'sphinx_multiversion',
-    'sphinx.ext.imgconverter'
+    'sphinx.ext.imgconverter',
+    'sphinx_copybutton'
 ]
 
 # Add any paths that contain templates here, relative to this directory.
```

And it should just work (tm).

closed 2023-06-04 by andreasWallner