SpinalHDL/SpinalDoc-RTD #152: Document compile SpinalHDL by mill.

Readon, 2022-11-20:
Do not merge until SpinalHDL/SpinalHDL#969 has not been merged.

numero-744, 2022-11-20:
Isn't it more related to SpinalTemplateSbt than to SpinalHDL? I think it is already supported? https://github.com/SpinalHDL/SpinalTemplateSbt/blob/master/build.sc

Readon, 2022-11-20:
> Isn't it more related to SpinalTemplateSbt than to SpinalHDL? I think it is already supported? https://github.com/SpinalHDL/SpinalTemplateSbt/blob/master/build.sc

It's about compile SpinalHDL itself by mill, not user's design.

numero-744, 2022-11-20:
For this section? https://spinalhdl.github.io/SpinalDoc-RTD/master/SpinalHDL/About%20SpinalHDL/faq.html#how-to-use-an-unreleased-version-of-spinalhdl-but-committed-on-git

Readon, 2022-11-20:
> For this section? https://spinalhdl.github.io/SpinalDoc-RTD/master/SpinalHDL/About%20SpinalHDL/faq.html#how-to-use-an-unreleased-version-of-spinalhdl-but-committed-on-git

Yes, almost, it's another way to speed up compiling time of the lib, while developing.
However, the official publishing can be remained.

commit 33244470 (2022-12-02): add docs for mill build. close #152
closed 2022-12-02 by Readon