SpinalHDL/SpinalDoc-RTD #157: Faq: publishLocal must match project's scala version

numero-744, 2022-11-24:
I did `sbt clean publishLocal` so it published for `2.11` but my project is using `2.12` so it didn't work.

Readon, 2022-11-24:
it depends on what do you want.
if you just want it once, just modify compiler's index in build.sbt.
if you want to deliver an argument to sbt to switch the compiler than more work is necessary.
also, if an auto detection is prefer than things get complicated.

numero-744, 2022-11-24:
Hmm I think we should not implement things, just document something like:

> If you are using scala xxxx then in `build.sbt` change xxx to xxx

or provide a command like:

`sbt clean '++ 2.12.13' publishLocal`

What do you think is better?

Readon, 2022-11-25:
> `sbt clean '++ 2.12.13' publishLocal`
> 
> What do you think is better?

This would be the better solution.

PR #158 (2022-12-02): Choosing scalaVersion with publishLocal
Closes #157 

Available on <https://numero-744.github.io/SpinalDoc-RTD/master/SpinalHDL/Introduction/faq.html#how-to-use-an-unreleased-version-of-spinalhdl-but-committed-on-git>
closed 2022-12-02 by Dolu1990