SpinalHDL/SpinalDoc-RTD #195: Structuring/parametrization.rst  ScopeProperty explanation

dlmiles, 2023-06-14:
what it does ?
in what situations is it better ?
example on how to use ?

Dolu1990, 2023-06-15:
Hi ^^

So, [ScopeProperty is](https://spinalhdl.github.io/SpinalDoc-RTD/master/SpinalHDL/Other%20language%20features/scope_property.html#scopeproperty) ?

> what it does ?

Mostly, it allows to have a thread local value, with some additional API to save the current context of all curent ScopeProperty and restore it later on.

> in what situations is it better ?

It allows to avoid global static variable, allowing to run multiple thread runnin at the same time while having different ScopeProperty values.
So, instead of global static variable => ScopeProperty

Also, it allows to apply a scope property to a portion of the code. It interpolate very well with the Fiber API. The current thread can halt for a while, then another thread can set its own scopeProperty value, when the initial thread will resume, it will still have its own value.

> example on how to use ?

```scala
object Xlen extends ScopeProperty[Int]

object ScopePropertyMiaou extends App{
  Xlen.set(1)
  println(Xlen.get) //1
  Xlen(2){
    println(Xlen.get) //2
  }
  println(Xlen.get) //1
}
```

dlmiles, 2023-08-27:
Ok I came back to address this one thinking I needed to convert the information in this issue into content for SpinalDoc-RTD but you have a link to the page for it.

**So I think the original issue in the documentation was missing :ref: links to the page which is not in place.**

source/SpinalHDL/Structuring/parametrization.rst:87
c549d2341d (Darryl L. Miles 2023-07-09 20:42:37 +0100  87) A :ref:`ScopeProperty <scopeproperty>` can also be used for configuration.

closed 2023-08-27 by dlmiles