SpinalHDL/SpinalDoc-RTD #205: Verilator C++ model compilation failed

ihaterecursion, 2023-08-04:
I have installed verilator 4.210 on win10 successfully ,and the way I install verilator is  compiling  source by git clone https://github.com/verilator/verilator. 
But a problem came out :
make: *** [VMyTopLevel.mk:77: VMyTopLevel__spinalWrapper.o] Error 1
make: *** [D:\program\msys\bin\mingw64\share\verilator/include/verilated.mk:239: VMyTopLevel__Trace.o] Error 1
make: *** [D:\program\msys\bin\mingw64\share\verilator/include/verilated.mk:245: verilated_vcd_c.o] Error 1
make: *** [D:\program\msys\bin\mingw64\share\verilator/include/verilated.mk:245: verilated.o] Error 1

Dolu1990, 2023-08-04:
Hi, did you followed those additional steps ? 
https://spinalhdl.github.io/SpinalDoc-RTD/master/SpinalHDL/Simulation/install/Verilator.html#windows

The error message isn't so nice, can you copy all the terminal message ?

Thanks :)

Dolu1990, 2023-08-04:
One thing could try to get a better error message, is to check with : 
https://github.com/SpinalHDL/SpinalHDL/blob/dev/sim/src/main/scala/spinal/sim/VerilatorBackend.scala#L448
updated to 

```scala
  class Logger extends ProcessLogger {
    override def err(s: => String): Unit = { if(!s.startsWith("ar: creating ")) println("ERR " + s) }
    override def out(s: => String): Unit = { if(!s.startsWith("ar: creating ")) println("OUT " + s) }
    override def buffer[T](f: => T) = f
  }
```

Dolu1990, 2023-08-07:
i pushed a similar change to the last post in SpinalHDL dev

Could you try with it ?  (https://spinalhdl.github.io/SpinalDoc-RTD/master/SpinalHDL/Introduction/faq.html#how-to-use-an-unreleased-version-of-spinalhdl-but-committed-on-git)

dlmiles, 2023-08-27:
Is this a documentation matter for the project SpinalDoc-RTD ?  It reads like a general support issue for SpinalHDL which should probably be over at https://github.com/SpinalHDL/SpinalHDL/issues/

@ihaterecursion  OP: Please consider closing this issue here if this logging update has resolved things.

andreasWallner, 2023-10-07:
@ihaterecursion did this solve your issues, can we close this issue?

closed 2023-10-09 by Dolu1990