SpinalHDL/SpinalDoc-RTD #274: GHDL simulation filed

VysochinAlexey, 2025-03-16:
sbt:projectname> runMain projectname.MyTopLevelSim
[info] running (fork) projectname.MyTopLevelSim 
[info] [Runtime] SpinalHDL v1.11.0    git head : 63852c61e498798f4e293594ce53fcb02c45eb6b
[info] [Runtime] JVM max memory : 3988,0MiB
[info] [Runtime] Current date : 2025.03.16 14:52:53
[info] [Progress] at 0,000 : Elaborate components
[info] [Progress] at 0,297 : Checks and transforms
[info] [Progress] at 0,413 : Generate VHDL to ./simWorkspace/tmp/job_1
[info] [Done] at 0,479
[info] [Progress] Simulation workspace in /home/src/MySpinalProject/./simWorkspace/MyTopLevel
[info] [Progress] GHDL compilation started
[info] [Progress] GHDL compilation done in 10,811 ms
[info] loading VPI module '/home/src/MySpinalProject/././simWorkspace/.pluginsCachePath/vpi_ghdl.vpi'
[info] VPI module loaded!
[info] Shared memory key : SpinalHDL_0_1_1225568095_-60726093_1742104374109_-7005297014291709098
[info] Start of simulation
[info] ../../src/ieee/v93/numeric_std-body.vhdl:1613:7:@0ms:(assertion warning): NUMERIC_STD."=": metavalue detected, returning FALSE
[info] [Progress] Start MyTopLevel test simulation with seed 699922504
[info] ../../src/ieee/v93/numeric_std-body.vhdl:1613:7:@0ms:(assertion warning): NUMERIC_STD."=": metavalue detected, returning FALSE
**[error] UNACCESSIBLE SIGNAL : (toplevel/zz_io_state :  UInt[8 bits]) isn't accessible during the simulation.**

MyTopLevel.vhd:
`signal` zz_io_state : unsigned(7 downto `0);`

MyTopLevel.scala:
`val` state = out UInt(8 bits)

Config.scala
`def` sim = SimConfig.withConfig(spinal).withGhdl

Dolu1990, 2025-03-16:
Hi, can you provide a full example to reproduce ?

VysochinAlexey, 2025-03-17:
**MyTopLevel.scala file:** from ..MySpinalProject/hw/spinal/projectname/

```scala
package projectname

import spinal.core._

// Hardware definition
case class MyTopLevel() extends Component {
  val io = new Bundle {
    val cond0 = in  Bool()
    val cond1 = in  Bool()
    val flag  = out Bool()
    val state = out unsigned(7 downto 0)
  }

  val counter = Reg(unsigned(7 downto 0)) init 0

  when(io.cond0) {
    counter := counter + 1
  }

  io.state := counter
  io.flag := (counter === 0) | io.cond1
}

object MyTopLevelVerilog extends App {
  Config.spinal.generateVerilog(MyTopLevel())
}

object MyTopLevelVhdl extends App {
  Config.spinal.generateVhdl(MyTopLevel())
}
```

**Config.scala file:**
```scala
package projectname

import spinal.core._
import spinal.core.sim._

object Config {
  def spinal = SpinalConfig(
    targetDirectory = "hw/gen",
    defaultConfigForClockDomains = ClockDomainConfig(
      resetActiveLevel = HIGH
    ),
    onlyStdLogicVectorAtTopLevelIo = true
  )

  def sim = SimConfig.withConfig(spinal).withGhdl
}
```
**MyTopLevelSim.scala file:**
```scala
package projectname

import spinal.core._
import spinal.core.sim._

object MyTopLevelSim extends App {
  Config.sim.compile(MyTopLevel()).doSim { dut =>
    // Fork a process to generate the reset and the clock on the dut
    dut.clockDomain.forkStimulus(period = 10)

    var modelState = 0
    for (idx <- 0 to 99) {
      // Drive the dut inputs with random values
      dut.io.cond0.randomize()
      dut.io.cond1.randomize()

      // Wait a rising edge on the clock
      dut.clockDomain.waitRisingEdge()

      // Check that the dut values match with the reference model ones
      val modelFlag = modelState == 0 || dut.io.cond1.toBoolean
      assert(dut.io.state.toInt == modelState)
      assert(dut.io.flag.toBoolean == modelFlag)

      // Update the reference model value
      if (dut.io.cond0.toBoolean) {
        modelState = (modelState + 1) & 0xff
      }
    }
  }
}
```

**MyTopLevel.vhd file:** from ..MySpinalProject/hw/gen/
```vhdl
entity MyTopLevel is
  port(
    io_cond0 : in std_logic;
    io_cond1 : in std_logic;
    io_flag : out std_logic;
    io_state : out std_logic_vector(7 downto 0);
    clk : in std_logic;
    reset : in std_logic
  );

end MyTopLevel;

architecture arch of MyTopLevel is

  signal zz_io_state : unsigned(7 downto 0);
  signal counter : unsigned(7 downto 0);
begin
  zz_io_state <= counter;
  io_flag <= (pkg_toStdLogic(counter = pkg_unsigned("00000000")) or io_cond1);
  io_state <= std_logic_vector(zz_io_state);
  process(clk, reset)
  begin
    if reset = '1' then
      counter <= pkg_unsigned("00000000");
    elsif rising_edge(clk) then
      if io_cond0 = '1' then
        counter <= (counter + pkg_unsigned("00000001"));
      end if;
    end if;
  end process;

end arch;
```

Dolu1990, 2025-03-27:
Hi,

this is related to     onlyStdLogicVectorAtTopLevelIo = true
Apparently, it isn't supported with the sim.

VysochinAlexey, 2025-03-28:
Remove  onlyStdLogicVectorAtTopLevelIo = true
```scala
package projectname

import spinal.core._
import spinal.core.sim._

object Config {
  def spinal = SpinalConfig(
    targetDirectory = "hw/gen",
    defaultConfigForClockDomains = ClockDomainConfig(
      resetActiveLevel = HIGH
    ),
  )

  def sim = SimConfig.withConfig(spinal).withGhdl
}
```
have got new errors:

sbt:projectname> runMain projectname.MyTopLevelSim
[info] compiling 2 Scala sources to /home/src/MySpinalProject/target/scala-2.13/classes ...
[error] /home/src/MySpinalProject/hw/spinal/projectname/MyTopLevel.scala:11:21: value unsigned is not a member of object spinal.core.out
[error]     val state = out unsigned(7 downto 0)
[error]                     ^
[error] /home/src/MySpinalProject/hw/spinal/projectname/MyTopLevel.scala:14:21: not found: value unsigned
[error]   val counter = Reg(unsigned(7 downto 0)) init 0
[error]                     ^
[error] two errors found
[error] (Compile / compileIncremental) Compilation failed
[error] Total time: 4 s, completed 28 мар. 2025 г., 15:26:49
sbt:projectname>

VysochinAlexey, 2025-04-01:
git clone --depth 1 https://github.com/SpinalHDL/SpinalTemplateSbt.git 
cd MySpinalProject
MySpinalProject/hw/spinal/projectname/  - The folder containing hardware descriptions MyTopLevel.scala file with "unsigned"

Dolu1990, 2025-04-01:
git clone --depth 1 https://github.com/SpinalHDL/SpinalTemplateSbt.git
cd SpinalTemplateSbt/hw/spinal/projectname
grep -r unsigned
=> nothing

VysochinAlexey, 2025-04-02:
MyTopLevel.scala with correct UInt types:
```scala
case class MyTopLevel() extends Component {
  val io = new Bundle {
    val cond0 = in  Bool()
    val cond1 = in  Bool()
    val flag  = out Bool()
    val state = out UInt(8 bits)
  }
  val counter = Reg(UInt(8 bits)) init 0
```

automatically vhdl-file create by sbt:
```vhdl
architecture arch of MyTopLevel is

  signal zz_io_state : unsigned(7 downto 0);
  signal counter : unsigned(7 downto 0);
begin
  zz_io_state <= counter;
  io_flag <= (pkg_toStdLogic(counter = pkg_unsigned("00000000")) or io_cond1);
  io_state <= std_logic_vector(zz_io_state);
```

[info] Start of simulation
[info] ../../src/ieee/v93/numeric_std-body.vhdl:1613:7:@0ms:(assertion warning): NUMERIC_STD."=": metavalue detected, returning FALSE
[info] [Progress] Start MyTopLevel test simulation with seed 371638831
[info] ../../src/ieee/v93/numeric_std-body.vhdl:1613:7:@0ms:(assertion warning): NUMERIC_STD."=": metavalue detected, returning FALSE
[info] [Error] Simulation failed at time=170
**[error] UNACCESSIBLE SIGNAL : (toplevel/zz_io_state :  UInt[8 bits]) isn't accessible during the simulation.**
[error] - To fix it, call simPublic() on it during the elaboration.
[info] End of simulation

attempt to add ``` modelState.simPublic() ```appear the new errors:

value simPublic is not a member of Int
[error]     modelState.simPublic()

Dolu1990, 2025-04-06:
Fixed with https://github.com/SpinalHDL/SpinalTemplateSbt/commit/976ceca3a31ebc0805f978eb08af749b575c9060.

GHDL simulation aren't supported with onlyStdLogicVectorAtTopLevelIo = true