SpinalHDL/SpinalDoc-RTD #275: simPublic tag is not working

VysochinAlexey, 2025-04-04:
```scala
import spinal.core.sim._

  class TopLevel extends Component {
    val counter = Reg(UInt(8 bits)) init(0) simPublic() // Here we add the simPublic tag on the counter register to make it visible
    counter := counter + 1
  }
```

[error] /SpinalTemplateSbt/hw/spinal/projectname/MyTopLevel.scala:15:44: can't supply unit value with infix notation because nullary method simPublic: (): spinal.core.UInt takes no arguments; use dotted invocation instead: spinal.core.sim.`package`.SimDataPimper[spinal.core.UInt](spinal.core.Reg.apply[spinal.core.UInt](core.this.HardType.implFactory[spinal.core.UInt](spinal.core.`package`.UInt(spinal.core.`package`.IntToBuilder(8).bits)), spinal.core.Reg.apply$default$2[Nothing], spinal.core.Reg.apply$default$3[Nothing]).init(spinal.core.`package`.IntToUInt(0))).simPublic()
[error]   val counter = Reg(UInt(8 bits)) init (0) simPublic()
[error]                                            ^
[error] one error found

Dolu1990, 2025-04-06:
That is a scala thing. Reg(UInt(8 bits)).init(0).simPublic() should be fine ?

VysochinAlexey, 2025-04-26:
```scala
 val counter = Reg(UInt(8 bits)) init (0)
```
working well

closed 2025-04-26 by VysochinAlexey